KumarRobotics/msckf_vio #131: Where can I find the IMU state in msckf_vio.cpp?

boeun98, 2022-04-01:
I want to compare msckf_vio.cpp with the paper "Robust Stereo Visual Inertial Odometry for Fast Autonomous Flight".

What represents the IMU state in msckf_vio.cpp that is same in the paper?

And I want to know what is child_frame_id in msckf_vio_euroc.launch.

I changed that to /zed/zed_node/odom. But the result is not good.

versatran01, 2022-04-01:
Child frame id is the child frame id of a transform. you can set it to whatever name you want. It does not affect the odometry result, so I don't know what you meant by "the result is not good"?
https://docs.ros.org/en/lunar/api/geometry_msgs/html/msg/TransformStamped.html

IMU state is at https://github.com/KumarRobotics/msckf_vio/blob/master/include/msckf_vio/imu_state.h

closed 2022-04-01 by versatran01

boeun98, 2022-04-04:
Thanks for your reply.

I used ZED camera and 3DMGX5-AHRS IMU of microstrain.

My rostopic list :
[image]
[image]
[image]

"the result is not good" is seen this video.

https://user-images.githubusercontent.com/72862584/161484525-30255262-d5e5-4440-8b80-1bc52d81778d.mp4

I modified 'child_frame_id' to `/zed/zed_node/odom` in msckf_vio_euroc.launch.

But, the  `/zed/zed_node/odom`  flied away, when I moved the ZED camera.

That is my important problem.

boeun98, 2022-04-05:
the ZED camera that I am using does not have IMU inside so I used an external IMU (3DMGX5-AHRS ) to use the topic in the package. Will it work if I use a static transformation between the base link of the ZED camera and the base link of IMU?