KumarRobotics/msckf_vio #134: about points in RViz

boeun98, 2022-04-25:
Hi, I want to know the points in RViz that are made if the msckf_vio launch file is implemented.
=============================================================
 my thought:

The points are the extracted features in the camera images.

and the points are used to calculate the length from the camera to the obstacle?

I don't know correctly what the points represent.
=============================================================
And, in my simulation, if the points aren't detected, the frame is filed away.

Please, help me.

Thank you.
.
.
.
.
.
<summary>

1. I want to know the points.

2.  in my simulation, if the points aren't detected, the frame is filed away. What is the problem?

fcladera, 2022-04-25:
The points that you are observing is are the extracted features that MSCKF is using. And indeed, if you are not tracking features, you may lose tracking. Unsure about what are the reasons for this, as it may be dependent on your simulation software. We have use MSCKF in simulated environments successfully.

closed 2022-04-25 by fcladera

boeun98, 2022-04-26:
Thx for your reply.

and I want to ask you something.

Do the point features represent the length between estimated points and the camera image frame?

Why the point features are needed?

fcladera, 2022-04-26:
The points are[ corners that are used to generate features](https://en.wikipedia.org/wiki/Features_from_accelerated_segment_test), which then are tracked between frames. This is the classic pipeline for sparse stereo vision. 

I recommend understanding some of the basics of CV for stereo vision, as it will help you understand better what this package does. https://www.coursera.org/learn/robotics-perception?specialization=robotics#syllabus