KumarRobotics/msckf_vio #137: about the covariance divergence

boeun98, 2022-06-04:
Thanks for your project.

I have a question.

When I did the VIO, the purple ellipsoid is bigger and bigger.

I think when there aren't the feature points, the purple ellipsoid is bigger and bigger.

It is the position covariance??? 

I want to know why the purple ellipsoid is bigger and bigger.

versatran01, 2022-06-04:
In vio XYZ and yaw are not observable so the their covariance will grow.
The covariance is simply extracted from the filter's full covariance matrix.

This could be caused by many things, for example inaccurate intrinsic/extrinsic calibration, wrong gravity norm, etc. It's hard to tell what exactly causes this.

closed 2022-06-04 by versatran01

boeun98, 2022-06-04:
Is the z-axis of the path getting smaller because the covariance also increases?

versatran01, 2022-06-04:
I don't understand your question

boeun98, 2022-06-05:
I'm talking about the figure above.

I experimented on the track (first figure).

But, the trajectory result is the second figure.

Is this result also due to increased covariance?
.
.
[image]
.
.
.
[image]

versatran01, 2022-06-05:
Could be, but most likely it's the other way around.

See details at https://en.wikipedia.org/wiki/Kalman_filter#Details