KumarRobotics/msckf_vio #149: run error for help

BoGuTU, 2024-05-13:
I tried to run this code but got this error
[image]
what is the reason for this？

LC-86886, 2024-06-05:
I have the same problem. Do you fix it? Could you message me If you solve it. Thanks.

BoGuTU, 2024-06-05:
> I have the same problem. Do you fix it? Could you message me If you solve it. Thanks.
yes, I solved it.
My problem was due to the opencv version, I reinstalled opencv 4.7.0 and ros ，then the programme worked.

LC-86886, 2024-06-05:
Thanks, It worked. But something different is that I make install opencv4.2 instead of opencv4.7. It also have the problem when I make install opencv4.7.